vvaltchev/tilck #75: False warning about absence of KVM on Arch Linux

ytret, 2021-08-01:
`scripts/templates/qemu/run_qemu` checks for presence of KVM in the system with `which kvm`:

https://github.com/vvaltchev/tilck/blob/3cde30be0f7d92760a66c99b677868ec822c4a2f/scripts/templates/qemu/run_qemu#L12-L20

I have KVM installed and the KVM module is loaded into the kernel:
```
$ lsmod | grep kvm
kvm_intel             335872  0
kvm                  1036288  1 kvm_intel
irqbypass              16384  1 kvm
```

But there is no `kvm` binary, so the check above fails.

The script also actually runs QEMU with `-enable-kvm` successfully and `info kvm` in the QEMU monitor outputs `kvm support: enabled`.

I understand that this is only a false warning, not a false error, but maybe there is another way of checking for absence of KVM that works on Arch Linux too?

vvaltchev, 2021-08-01:
Good point.
What about checking KVM's presence with: ` if lsmod | grep kvm &> /dev/null; then` ?

ytret, 2021-08-01:
Grepping only `kvm` is not enough, because for KVM to work properly either `kvm_intel` or `kvm_amd` has to be loaded. For example, on my machine QEMU errors with `failed to initialize kvm` without `kvm_intel`. So it's reasonable to write:

```
if ! (lsmod | grep kvm_amd &> /dev/null) && ! (lsmod | grep kvm_intel &> /dev/null); then
```

Regarding your last question, that's not my case, at least. I have just rebooted, run `lsmod | grep kvm` before running QEMU and the modules are there. I don't have any `/etc/modprobe.d` rules or any `modprobe`-ing scripts, hence the modules were loaded during boot thanks to some hardware detection.

vvaltchev, 2021-08-01:
Mmm... isn't checking for either `kvm_amd` or `kvm_intel` a bit too tight as a requirement? If tomorrow the KVM guys create a new kvm_generic_XYZ module, those checks will fail and we'll see a false positive again. On the other hand, if we are in the corner case described here: https://wiki.archlinux.org/title/KVM:

> Tip: If modprobing kvm_intel or kvm_amd fails but modprobing kvm succeeds, and lscpu claims that hardware acceleration is supported, check the BIOS settings. Some vendors, especially laptop vendors, disable these processor extensions by default. To determine whether there is no hardware support or whether the extensions are disabled in BIOS, the output from dmesg after having failed to modprobe will tell.

We'll have `kvm` loaded but neither `kvm_amd` or `kvm_intel`, because KVM is installed but the loading of kvm_amd or kvm_intel failed, we won't see a warning and QEMU -enable-kvm will fail with a different error. I believe that's acceptable as it's unlikely to happen and is not related with the installation of KVM per se, but with other things like BIOS settings.

vvaltchev, 2021-08-01:
Merged. Thanks for the fix!

PR #76 (2021-08-01): [scripts] Check for presence of KVM with `lsmod | grep kvm`
Fixes #75
commit 65c7ca6f (2021-08-01): [scripts] Check for presence of KVM with `lsmod | grep kvm`

Fixes #75
closed 2021-08-01 by vvaltchev