vvaltchev/tilck #79: Building fbdoom

johanburati, 2021-08-04:
I am getting an error when I try to build **fbdoom** on Ubuntu 20.04 x86_64.
Maybe I am missing some 16bit libraries, does anyone knows which ones I need to install ?

```
$ ./scripts/build_toolchain -s build_fbdoom

ARCH=i386
GCC_TC_VER=7.3.0
ARCH_GCC_TC_VER=stable-2018.11-1

CI Detected: None [Assuming user machine]
[build_toolchain] Executing single function 'build_fbdoom'
NOTE: Skipping FBDOOM
[build_toolchain] Executing single function: DONE

$ cat ./toolchain2/i386/fbDOOM/fbdoom/build.log
mkdir -p build
[Compiling i_video_fbdev.c]
[Compiling i_input_tty.c]
[Compiling i_main.c]
[Compiling dummy.c]
/tmp/tilck/scripts/build_scripts/i686-linux-gcc: 2: exec: -m32: not found
make: *** [Makefile:61: build/i_video_fbdev.o] Error 127
/tmp/tilck/scripts/build_scripts/i686-linux-gcc: 2: exec: -m32: not found
make: *** Waiting for unfinished jobs....
make: *** [Makefile:61: build/i_input_tty.o] Error 127
/tmp/tilck/scripts/build_scripts/i686-linux-gcc: 2: exec: -m32: not found
make: *** [Makefile:61: build/i_main.o] Error 127
/tmp/tilck/scripts/build_scripts/i686-linux-gcc: 2: exec: -m32: not found
make: *** [Makefile:61: build/dummy.o] Error 127
```

vvaltchev, 2021-08-04:
Hi!
I'm unable to reproduce your issue, not even on a fresh install of Ubuntu 20.04. `build_fbdoom` uses a prebuilt GCC i686 toolchain downloaded the first time `./scripts/build_toolchain` is run, so there's no need to install 32-bit cross-compilers at system level.

Did you run `./scripts/build_toolchain` without options first? That's the first step required after cloning Tilck's repo:
https://github.com/vvaltchev/tilck#building-tilck

Or, if you're interested in the long story, there's also this document: https://github.com/vvaltchev/tilck/blob/master/docs/building.md

After that, of course, you can install additional packages using the `-s` option. 

Let me know if this solves your problem :-)

Have a nice day,
Vlad

johanburati, 2021-08-04:
Hi Vlad, thank you I've tried again and could build it this time, maybe I missed running the script without the *-s option*.
Now I need to figure out how to get the framebuffer thing to work with **qemu**, I am getting a `Could not open /dev/fb0` error when trying to run **fbdoom**, any idea how to fix that ?

vvaltchev, 2021-08-05:
I'm happy to help :-)
You could get such an error only if you booted Tilck in "text mode". The bootloader allows you to choose text mode or plenty of other resolutions. 800x600 is the default one. 

If you're using Linux, just run `./build/run_qemu` and boot using the default settings by just pressing ENTER.

If you're using WSL, running Linux graphical applications (such as QEMU) is more tricky. You'd need to install an X server on Windows and then configure properly the DISPLAY environment variable in WSL. You should probably google for a generic tutorial about that, as it has nothing to do with Tilck. BTW, at some point, when WSLg will become mainstream, it would be much easier to run Linux graphics apps from WSL.

johanburati, 2021-08-06:
Awesome it's working ! I was using the kernel and the ramdisk, I switched to using the image ;-)

vvaltchev, 2021-08-06:
That’s great!

closed 2021-08-06 by vvaltchev