vvaltchev/tilck #99: ACPI related problems

duselguy, 2022-06-08:
I included few different problems in one issue, because (may be) they relate to the same problem.
----------------------------------------------------------------------------------------------------------------------------------
**Describe the bug**
ACPI related message appears only once.

**Host configuration**
Amilo UI 3520

**Toolchain configuration**
Standard

**Tilck build configuration**
See attachment below

**Type of issue (select one)**
- Incorrect behavior

**Reproduction details**
1. Boot with power cable is ON
1.1. Switch power cable OFF after the boot completed
1.2. The message: 

> ACPI: got GPE #29

1.3. Switch power cable ON/OFF/ON/etc. - no above message (not trapped or logged)
2. Boot with power cable is OFF
2.1. Switch power cable ON after the boot completed
2.2. The message: 

> ACPI: got GPE #29

2.3. Switch power cable OFF/ON/OFF/etc. - no above message (not trapped or logged)

**Expected behavior**
The message after each switch of the power cable.

----------------------------------------------------------------------------------------------------------------------------------
**Describe the bug**
poweroff command is not completed

**Host configuration**
Amilo UI 3520

**Toolchain configuration**
Standard

**Tilck build configuration**
See attachment below

**Type of issue (select one)**
- Incorrect behavior

**Reproduction details**
1. Boot. Than issue: poweroff
1.1. Executing, black screen, but power and WiFi indicators are light.
2. Boot. Than issue poweroff -f
[image]

**Expected behavior**
Black screen and no light indicators.
----------------------------------------------------------------------------------------------------------------------------------
**Describe the bug**
ACPI related message with **exact number** appears only once.

**Host configuration**
Lenovo G410

**Toolchain configuration**
Standard

**Tilck build configuration**
See attachment above

**Type of issue (select one)**
- Incorrect behavior

**Reproduction details**
1. poweroff and poweroff -f work OK
2. After the boot:

> ACPI: got GPE #2 (?)

> ACPI: got GPE #27 (closing the screen cover)

Closing the screen cover again - no ACPI message

**Expected behavior**
The message after each closing the screen cover.

vvaltchev, 2022-06-08:
OK, ACPI power events are not really supported at the moment. So, I'm not surprised to see the GPE appear inconsistently. I never had enough time to spend on this. The lid open/close events don't work.

In my understanding, the problem is that Tilck does not implement the address space for the embedded controller (EC). Without it, many things don't work properly.

The poweroff instead, should work.

vvaltchev, 2022-06-08:
Just one thing, could you try to take photos in focus? I know it's a minor detail, but text is hard to read :-)

duselguy, 2022-06-08:
Sorry, my phone is too old to get good shots. Will use a separate camera next time.

duselguy, 2022-06-11:
Tried Alpine Linux on Amilo book (see above). poweroff and poweroff -f work ok (black screen and no led light).

vvaltchev, 2022-06-11:
Not surprised, Linux has a huge HW support. It runs on almost anything and has a huge base of beta-testers. Tilck is mostly tested on just the machines I personally own :-)

Anyway, yeah, the poweroff not working is definitively a Tilck bug. The other things related to GPE events are in the "not yet implemented" category.

vvaltchev, 2022-06-12:
@duselguy I didn't have time to look at the bug with the ACPI poweroff, but I noticed that `poweroff -f` and `reboot -f` didn't behave properly due to a small bug of mine and could get stuck. Again, this doesn't fix this ACPI-related bug, but it's something.

duselguy, 2022-06-12:
Thank you,
1) poweroff, poweroff -f, reboot, reboot -f work as expected now.
2) The problem with ACPI/POWER/LED (see above) is left until better times.
Regards,
Vladimir

vvaltchev, 2022-06-12:
That's great, thank you Vladmir!!
I'm closing this issue because what mattered was to get poweroff and reboot to work on your machine too.
The rest is in a long TODO list :-)

closed 2022-06-12 by vvaltchev

duselguy, 2022-06-19:
Investigation Power/Led problem on Amilo UI 3520 (To Be Continued):
1) Typo in osl_misc.c (AcpiOsEnterSleep):
`  printk("ACPI sleep: %u, 0x%x, 0x%x\n", SleepState, RegaValue, RegaValue);`

vvaltchev, 2022-06-23:
That’s interesting. I missed this update.
So poweroff works, but just the leds might remain ON depending on their previous state, right?

duselguy, 2022-06-24:
@vvaltchev
Almost so :-)

**Scenarios:**
1) Press the power button. Both indicators are on: power and Wi-Fi. We boot and issue the poweroff command: the screen goes blank, but both indicators (power and Wi-Fi) remain on.
2) Press the power button. Only the power indicator is on. We boot and issue the poweroff command: the screen and the power indicator go out.

**Trick:**
To execute scenario 2), you must forbid WI-Fi ability before booting using the function key combination (after that the Wi-Fi indicator is off).

**Conclusion:**
It looks like in scenario 1) the current ACPI related code (after poweroff command) is not enough to really power off the netbook - both indicators (power and Wi-Fi) remain on.

vvaltchev, 2022-06-24:
@duselguy Thanks for the investigation, Vladimir. The whole picture makes perfectly sense. In order to function properly on all devices, poweroff requires first powering off some ACPI devices, like the Wi-Fi network device in your case. Tilck does nothing like that at the moment and completely relies on ACPI to do everything automatically. Unfortunately, it's not that simple sometimes. 

I'll keep that in mind as a future TOD task.

duselguy, 2022-06-24:
@vvaltchev
I don't quite understand the purpose/behavior of the WiFi indicator on this netbook. Usually this LED should be on if we have a WiFi connection. In our case, it is also used to show whether WiFi connection is possible or not.
**Summary:**
1. Amilo
1.1. Tilck: The behavior of the _poweroff_ command is described in my previous comment.
1.2. Alpine Linux: _poweroff_ command turns off both LEDs (power and WiFi).
2. Lenovo (has a physical switch, if it is turned on, the corresponding WiFi indicator is off, and this is understandable).
2.1. Tilck:
The _poweroff_ command turns off the power LED and is independent of the state (on/off) of the WiFi physical switch.

vvaltchev, 2022-06-24:
@duselguy The WiFi indicator typically shows if the WiFi network adapter is turned on. That was nothing to do with the WiFi connection. The hotkey combination that you discovered allows you to turn on/off your WiFi hardware.

The purpose of that is AirPlane mode etc.

duselguy, 2022-06-24:
@vvaltchev
Lenovo:
1. Set the physical switch to OFF.
2. Boot Windows
[image]
3. Set the physical switch to ON. See the photo above.
4. Connect WiFi
[image]
5. Disconnect WiFi
[image]

Let this be the last attempt, because for now it doesn't matter -:).
Best regards,
Vladimir