vvaltchev/tilck #101: Memory leak in compiled GCC.

turbocat2001, 2022-06-21:
Builded the toolchain with a script. Changed it a bit to use pacman for my Artix Linux. The toolchain is assembled and the package is installed. However, when the kernel compilation hits 53 percent, I run out of memory and Xorg killed.

vvaltchev, 2022-06-22:
Hello :-) I've observed the same kind of problem while building any big-enough project on machines which a low amount of usable memory. Simply, by using `make -j`, GNU make spawns as many GCC instances as possible and each one consumes a fair amount of memory (hundreds of MBs): at some point, your system runs out of memory and the OOM killer is run: it selects a process to kill according to a complex system of heuristics. In your case, it chose Xorg. That has nothing to do with Tilck, nor it's build system or GCC. And it's definitively not a memory leak.

Therefore, if you don't have enough RAM, just build with `make` or at most `make -j2`. Limit the level of parallelization.

closed 2022-06-22 by vvaltchev

turbocat2001, 2022-06-22:
Thank you!  I've just never encountered it.  Sorry.

duselguy, 2022-06-23:
Suddenly have almost the same on the build with make -j (Ubuntu freezes, hard reboot). The log shows zero free swap space. Moved to the simple make :-)

vvaltchev, 2022-06-24:
@duselguy how much RAM do you have? try at least make -j2

duselguy, 2022-06-24:
@vvaltchev
4GB RAM and 2GB swap. At the build time with _make -j_ I had Chrome opened with GitHub tab. It looks like Chrome failed (something like oops on Chrome page appeared + warnings about Chrome extensions crashes). Ok, but what I don't like is Ubuntu freezing (keyboard and mouse are blocked) at out of memory.

vvaltchev, 2022-06-24:
Unfortunately, that's pretty much inevitable because of the way the whole software stack is built (it's complicated). I don't like that we (the software industry) ended up in this position. For other operating systems is more or less the same. Just, the swap might dynamically expand on the fly but, if it reaches its hard limit, the whole system will loose stability because doesn't know which process must be killed.

Anyway, the easiest solution is to just increase the size of the swap or to get more RAM.