vvaltchev/tilck #118: Compile-out with #ifdef print_banner_line() and show_tilck_logo() when KERNEL_SHOW_LOGO=0

vvaltchev, 2022-08-08:
As it can be seen from https://app.codecov.io/gh/vvaltchev/tilck/blobs/master/kernel/misc.c, `print_banner_line()` and `show_tilck_logo()` don't have coverage, despite being always run on boot. That's simply because in coverage builds, the env variable `TILCK_NO_LOGO` is set. See https://github.com/vvaltchev/tilck/blob/master/other/coverage_build.yml. 

Because those trivial code lines that don't really need automated testing (we always see their output on boot) and we don't care about their coverage, it would nice to just compile them out when TILCK_NO_LOGO is set. Doing that is pretty simple, considering that setting the env variable TILCK_NO_LOGO does nothing more than setting the default value of the CMake option `KERNEL_SHOW_LOGO`, which is already exported in `<tilck_gen_headers/mod_console.h>`.

Therefore, the fix would require just to use surround `print_banner_line()` and `show_tilck_logo()` with `#if KERNEL_SHOW_LOGO` and `#endif`, making sure there are no compile errors.

lvntky, 2022-08-13:
Hi @vvaltchev, I surround the `print_banner_line()` and `show_tilck_logo()` with the `#if KERNEL_SHOW_LOGO` and `#endif`. I build the tilck and run on qemu also build and test with gtests, didn't occure with any error. Also i tkinter with `mod_console.h` a bit, I changed the `#define    KERNEL_SHOW_LOGO 1` to 0(for checking the #if KERNEL_SHOW_LOGO works on misc.c) and GCC gives error for `show_hello_message()` because show_hello_message needs `show_tilck_logo();` and `show_tilck_logo();` did not implemented because i set the `KERNEL_SHOW_LOGO 0` on the `mod_console.h` , so I think it works. I dind't open up a pull request because I wanted to say it to you first and be sure I didn't do anything wrong :)

PS: I set back the  `#define    KERNEL_SHOW_LOGO 1` as default after the test

vvaltchev, 2022-08-17:
Hello @lvntky, thanks for working on that. Sure, your plans sounds OK. The change should consist just in adding an `#ifdef` + an `#endif`. But, if you wanna check if everything compiles properly in both cases (which is necessary), touching `mod_console.h` is not the right thing to do. (Also note: there are two mod_console.h files: the template one in `config` and the automatically generated one in a subdirectory of the build directory.)

You should install `cmake-curses-gui` and then run `ccmake build`. That tool has a simple curses UI which will allow you to re-configure the Tilck compile-time options. Toggle the `KERNEL_SHOW_LOGO` option from ON to OFF, press `c` to re-configure and then `g` to re-generate the makefiles and the other objects like the generated headers (including mod_console.h). After that, run a build with `make`. Check that the build passes when the option is ON as well as when the option is OFF. If the build fails in case the `KERNEL_SHOW_LOGO` is OFF, you should figure out a way to fix it. It should be super trivial in any case.

Sorry for now responding quickly, but as I said, I'm currently on vacation with very limited internet connection and time. I guess this should be enough for you to prepare the change and open a pull request. I'll review it as soon as I come back at the end of the week.

lvntky, 2022-08-17:
hi @vvaltchev , thank you for these great information, I will prepare a pull request in light of this information. 
Yes I know you are on vacation, I hope you're having a good one :)  See you after the vacation

vvaltchev, 2022-08-18:
This issue has been fixed by pull request https://github.com/vvaltchev/tilck/pull/126

closed 2022-08-18 by vvaltchev