vvaltchev/tilck #120: Write a unit test to trigger the error code path in fork_dup_all_handles()

vvaltchev, 2022-08-08:
As it can be seen here: https://app.codecov.io/gh/vvaltchev/tilck/blobs/master/kernel/fork.c, currently the following code path inside `fork_dup_all_handles()` is not covered by tests:

```C
      if (rc < 0 || !dup_h) {

         enable_preemption();
         {
            for (u32 j = 0; j < i; j++)
               vfs_close(pi->handles[j]);
         }
         disable_preemption();
         return -ENOMEM;
      }
```

Now that we have gMock in Tilck, it would be relatively simple to just mock `vfs_dup` and `vfs_close()` so that inside a unit test we can trigger this code path. To do that, it would be necessary to make the `fork_dup_all_handles()` function `STATIC` instead of `static`, so that in unit tests, the function won't be static anymore. Also, it would be necessary add the mocked functions to a CMakeLists.txt file and to another file in the unit tests directory. For the exact instructions, see the comments in: https://github.com/vvaltchev/tilck/blob/master/tests/unit/mock_experiments.cpp

Clearly, because the `fork_dup_all_handles()` function takes a `struct process *` parameter, the unit test will have to allocate such a structure and set at least 2 handles in the `pi->handles` array.

michelececcacci, 2022-09-18:
Even though i'm not a first time contributor i would like to take this.

vvaltchev, 2022-09-18:
@michelececcacci Sure, take the issue. You haven't contributed in this area, so I guess it's still a very good first task for you.

closed 2022-10-10 by vvaltchev