vvaltchev/tilck #121: Move kernel/fs/fat32_debug.c to tests/unit/

vvaltchev, 2022-08-08:
The file contains just debug-util functions to dump a FAT32 in-memory partition. It doesn't make sense to keep it in the kernel itself.
Note: its declaration in `include/tilck/common/fat32_base.h` has to be removed as well.

lvntky, 2022-08-09:
hi @vvaltchev i want to contribute the tilck and think i can handle this, can you assign me to the issue? Or may I be a contributer?

vvaltchev, 2022-08-09:
Hi @lvntky, you’re absolutely welcome to contribute. I’ve assigned the issue to you.

lvntky, 2022-08-09:
Thank you very much @vvaltchev ! Even though I am new to the contributing OSS, I really like tilck, I will do my best and hopefully be a helpful long-term contributor. Tomorrow I will start with reading the contribution and testing guidelines!

vvaltchev, 2022-08-11:
After merging https://github.com/vvaltchev/tilck/pull/125, this issue has been fixed.

closed 2022-08-11 by vvaltchev