vvaltchev/tilck #135: Move other/musl-gcc and other/musl-g++ to scripts/templates

vvaltchev, 2022-09-09:
The `other/musl-gcc` and `other/musl-g++` template scripts are currently in the wrong place. The fix requires simply those two files to `scripts/templates` and replacing their paths in the `other/cmake/gen_config_pre.cmake` file:

Code snippet:

```cmake
smart_config_file(
   ${CMAKE_SOURCE_DIR}/other/musl-gcc
   ${MUSL_GCC}
)

smart_config_file(
   ${CMAKE_SOURCE_DIR}/other/musl-g++
   ${MUSL_GXX}
)
```

francescoparra, 2022-09-25:
Hello, may I take this issue?

vvaltchev, 2022-09-25:
Hello @francescoparra. Sure, absolutely :-)

vvaltchev, 2022-09-26:
Closing this issue, as it has been fixed by commit https://github.com/vvaltchev/tilck/commit/32b571d5625c04eefb9bf7a3ba732dc6c1e2cf53

closed 2022-09-26 by vvaltchev