vvaltchev/tilck #142: Contribution documentation is not easily found

francescoparra, 2022-09-25:
When I had the first approach with the project I was searching for the `contributing.md` file without seeing the `docs` folder.

I think that move the file to the main folder would be better.

Another alternative could be to speicify the file in the `README.md`

vvaltchev, 2022-09-25:
Thank you for the feedback, @francescoparra.
Originally, I introduced the `contributing.md` file in the main folder along with `code_of_conduct.md`, but it turned out that I didn't like it because the main folder started to have too many entries. I don't like it too crowded, in particular with long file/directory names. So, I moved both the files to `docs/`, to make it more clean.

I guess the best should be at this point to mention the file in the README.md.

vvaltchev, 2022-09-25:
I added a short contributing section in the README file.

closed 2022-09-25 by vvaltchev