vvaltchev/tilck #144: Warn not to mock and wrap the same function in mocking_experiments.cpp

michelececcacci, 2022-10-11:
I feel like this would be a better experience for people that are not well versed on c++(possibly first time contributors). If you feel like it's not necessary, just close the issue

vvaltchev, 2022-10-11:
I'm not sure what do you mean. Did you add originally a __wrap function? I believe the instructions in the file are CLEAR:

```C
/*
 * Mocking of C functions in the kernel
 *
 * Instructions:
 *
 *    1. add all the functions that we would like to mock in the WRAPPED_SYMS
 *       list in tests/unit/CMakeLists.txt
 *
 *    2. add all of those functions in tests/unit/mocked_funcs.h, with their
 *       correct signature
 *
 *    3. create gMOCK classes with one or more mocked "methods" like MockingBar
 *
 *    4. instantiate ONE mock object per TEST and use it with EXPECT_CALL or
 *       ON_CALL, as explained in the gmock documentation
 */
```

Is there any step missing in the instructions? 
We cannot warning people against any kind of possible mistake they could possibly make.

michelececcacci, 2022-10-11:
The instructions are really clear, i just didn't want anyone to repeat a mistake i did (wrapping and mocking a function at the same time)

closed 2022-10-11 by michelececcacci