vvaltchev/tilck #151: Unable to build on Ubuntu 21.10

malisipi, 2022-11-29:
**Describe the bug**
Unable to build on Ubuntu 22.10.

**Host configuration**

**ARCH:** i386
**HOST_ARCH:** x86_64
**GCC_TC_VER:** 8.4.0
**ARCH_GCC_TC_VER:** stable-2020.02-2

**Toolchain configuration**
I used only `./scripts/build_toolchain`

**Tilck build configuration**
I used only `make` command only. 
Commit: 2cb2bf7edfe15cd880e2a42a842ddbb50412601d

**Type of issue (select one)**
- Toolchain build problem

**Reproduction details**

- Is the issue reproducible in a deterministic way? Yes

**Expected behavior**
`tilck.img` never builds.

**Additional context**
I attached my terminal output.
[tilck_output.txt](https://github.com/vvaltchev/tilck/files/10115287/tilck_output.txt)

vvaltchev, 2022-11-29:
Hello @malisipi,
from the attached log file I see the following error:

```
[ 99%] Creating a dual-bootable (legacy+efi) image file (MBR)
  /home/malisipi/tilck/tilck/toolchain2/host_x86_64/mtools/mmd: File "usr/share/terminfo/l" not found
make[3]: *** [CMakeFiles/tilck_image.dir/build.make:94: tilck.img] Error 1
make[3]: *** 'tilck.img' dosyası siliniyor
make[2]: *** [CMakeFiles/Makefile2:318: CMakeFiles/tilck_image.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
make: *** [Makefile:13: all] Error 2
```
and it looks weird that it cannot find `usr/share/terminfo/l` as the directory is in `sysroot`. Could you please attach the output of:
```
ls -Rl sysroot
```
run on Tilck's root directory?

Also, could you please attach the output of the following three commands?
```
ls -lh  toolchain2/host_x86_64/mtools/mmd
file toolchain2/host_x86_64/mtools/mtools
./toolchain2/host_x86_64/mtools/mtools
```

Thank you,
Vlad

malisipi, 2022-11-29:
Thank you for attention. I ran these commands. But it seems everything is correct.

```
> ls -Rl sysroot
sysroot:
toplam 16
drwxrwxr-x 2 malisipi malisipi 4096 Kas 29 23:43 bin
drwxrwxr-x 2 malisipi malisipi 4096 Kas 29 23:43 etc
-rw-rw-r-- 1 malisipi malisipi  139 Kas 29 23:43 README
drwxrwxr-x 5 malisipi malisipi 4096 Kas 29 23:43 usr

sysroot/bin:
toplam 0

sysroot/etc:
toplam 12
-rw-rw-r-- 1 malisipi malisipi  10 Kas 29 23:43 group
-rw-rw-r-- 1 malisipi malisipi  27 Kas 29 23:43 passwd
-rw-rw-r-- 1 malisipi malisipi 937 Kas 29 23:43 start

sysroot/usr:
toplam 12
drwxrwxr-x 2 malisipi malisipi 4096 Kas 29 23:43 bin
drwxrwxr-x 2 malisipi malisipi 4096 Kas 29 23:43 lib
drwxrwxr-x 3 malisipi malisipi 4096 Kas 29 23:43 share

sysroot/usr/bin:
toplam 4
-rwxrwxr-x 1 malisipi malisipi 1006 Kas 29 23:43 vim

sysroot/usr/lib:
toplam 0

sysroot/usr/share:
toplam 4
drwxrwxr-x 3 malisipi malisipi 4096 Kas 29 23:43 terminfo

sysroot/usr/share/terminfo:
toplam 4
drwxrwxr-x 2 malisipi malisipi 4096 Kas 29 23:43 l

sysroot/usr/share/terminfo/l:
toplam 4
-rw-rw-r-- 1 malisipi malisipi 1822 Kas 29 23:43 linux
```

```
> ls -lh  toolchain2/host_x86_64/mtools/mmd

lrwxrwxrwx 1 malisipi malisipi 6 Kas 29 23:45 toolchain2/host_x86_64/mtools/mmd -> mtools

> file toolchain2/host_x86_64/mtools/mtools

toolchain2/host_x86_64/mtools/mtools: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=9243031768f481026db5a5db5e2282916ea642f2, for GNU/Linux 3.2.0, not stripped

> ./toolchain2/host_x86_64/mtools/mtools

Supported commands:
mattrib, mbadblocks, mcat, mcd, mclasserase, mcopy, mdel, mdeltree
mdir, mdoctorfat, mdu, mformat, minfo, mlabel, mmd, mmount
mpartition, mrd, mread, mmove, mren, mshowfat, mshortname, mtoolstest
mtype, mwrite, mzip
```

vvaltchev, 2022-11-30:
Hello @malisipi,
from what I see, everything is OK. 

Tilck for sure _does_ build on Ubuntu 21.10 in the general case as Ubuntu is the main development distro. There's something weird with your setup. The (generated) script that fails should be: `./build/scripts/build_fatpart`. Try running it manually and debugging it a bit. The only place `mmd` is used is in the `make_fatpart` function:

```Bash
function make_fatpart {

   create_fatpart_if_necessary
   format_fatpart

   # first, create the directories
   for f in $(find * -type d); do
      $mmd -i $dest $f                   # <<< HERE
   done

   # ...
```
Maybe try adding statements like `pwd` before the for loop. Also, check that the fatpart file in `.build/fatpart` do exist. Run `mdir` to check its contents with:
```
$ ./toolchain2/host/mtools/mdir -i ./build/fatpart
```

Btw, long shot, but it's possible that's a weird localization issue. I saw from your logs that you don't use "EN_us" or "C".

malisipi, 2022-11-30:
> Btw, long shot, but it's possible that's a weird localization issue. I saw from your logs that you don't use "EN_us" or "C".

I have created a new user that uses English as primary language. And everything works fine now. Thanks you too much.

vvaltchev, 2022-11-30:
You're welcome. Probably the bug is in mtools.