vvaltchev/tilck #155: Distro required

simon1tan1, 2023-02-08:
I think this is a great project. IMHO, I think in order for adoption, a distro with a package manager would be nice for people to use and play with, see what software works, what does not. "kick the tires". That way people don't have to start from scratch. I know you're primary aim is towards system programmers who can help you with the kernel but there are probably just as many non system programmers who want to see something like this project thrive. Just a thought.

vvaltchev, 2023-02-08:
Hello Simon,
I'm glad that you like the project. 

What do you mean exactly with "a distro with a package manager" ? Do you mean that would you like to install Tilck with `apt install <proj-name>` ? 

I'll try to explain why I don't believe that's a good idea after all (if I've understood correctly your question).

* First of all, Tilck is **not** an application: currently it cannot be used for any meaningful job. It can be used only by developers to further extend it. Therefore, having it as a binary package installable on major distros is pointless. It doesn't matter your level of experience, you won't be able to do anything practical with it. It only makes sense to download its source and work on it.

* Second, Tilck has one of most user-friendly scripts for downloading and installing its requirements I've ever seen on any open source project. Making that super easy was one of my goals. Try yourself to download and compile any other educational operating system: https://wiki.osdev.org/Projects and compare your experience with Tilck.

* Third, even if Tilck was more mature and usable for something practical it's still **not** an application. Let's for a moment compare it with other operating systems: can you install a FreeBSD VM on Ubuntu with `apt install freebsd` ? Of course not. Can you possibly install Windows that way? Nope. Neither Linux VMs can be installed on other operating systems, with the special exception of WSL 2, but that's completely different and has a good purpose. Operating systems are not generally available as applications for other operating systems.

Even in the long-term, the project is meant to run on ARM and/or Risc V SoC, not on your typical machine. It's not meant for people to play DOOM or Tetris on it. I added support for those games after people pushing for that, just as a show-off. 

The closest thing I can think of that might make you happy is for me to publish a pre-built ISO image of Tilck, with instructions about how to run it on QEMU on Linux. That would be certainly possible, and maybe is a good idea. But, that won't stimulate people to try building it from source, which is an essential part of the "developer experience"  :-( 

But I'll think about that.

Thank you for the feedback,
Vlad

simon1tan1, 2023-02-09:
I'm thinking your project is a kernel so the thought was to build a distribution around it sort of how Debian, Redhat, Alpine is built around Linux. Yes, by package manager, like apt install <package>, I don not mean install Tilck like that. I do understand Tilck is not an application that can be installed just with with package manager.

I did not realize the goal is for ARM/RISCV. I see x86 mentioned many time in the README so was under a  different assumption. Please do realize ARM and RISCV will not always be relegated to embedded systems(see Mac, Ampere, Graviton). Knowing you are aiming this at embedded systems, a distro may not the best idea.

Still, keep up the good work. 
 
Cheers!

closed 2023-02-09 by simon1tan1

hatonthecat, 2023-02-09:
i can see both viewpoints, as I have an interest in both of these ideas and projects. While Tilck certainly would need more development for it to be a ready-made distro, as the developer points out, it is still primarily for developers to test and extend. 

That said, if it did reach the level where it could be a practical "drop-in" replacement for other kernels so that it were binary compatible, it would still need some maintenance by, kernel developers I presume.

I think an easier way to understand Tilck would be to compare it to Tiny Core Linux, which also does not intend to see itself as a turn-key distribution: http://distro.ibiblio.org/tinycorelinux/intro.html

https://tiny.slitaz.org/index.php is a web-based package manager, which I think is an interesting model for potential distros.

vvaltchev, 2023-02-09:
Sorry for @simon1tan1 for completely misunderstanding your comment. Thanks @hatonthecat for the comments: the comparison to Tiny Core Linux is a good one. 

Yeah, I don't plan Tilck to ever be the right kernel for a distro. Even for embedded, I plan to make it as simple as possible: all the software will be loaded in-memory from a partition. If we get to point where updates from the internet will be required, the system download the new data & just reboot in less than a second. Desktop & server operating systems are waaaay more complex to support.

For people needing the actual power of a Linux distro, I'd warmly recommend using Linux itself. There's no way a project like Tilck to reach that amount of features and even if it could, it would be a waste trying to re-implement the wheel. I'd like to write something small, simple & extra-reliable, not a big and complex operating system.

PC-Doctor666, 2024-02-01:
> I'm thinking your project is a kernel so the thought was to build a distribution around it sort of how Debian, Redhat, Alpine is built around Linux. Yes, by package manager, like apt install , I do not mean install Tilck like that. I do understand Tilck is not an application that can be installed just with a package manager.
> 
> I did not realize the goal is for ARM/RISCV. I see x86 mentioned many time in the README so was under a different assumption. Please do realize ARM and RISCV will not always be relegated to embedded systems(see Mac, Ampere, Graviton). Knowing your are aiming this at embedded systems, a distro may not the best idea.
> 
> Still, keep up the good work.
> 
> Cheers!

If this ever happens in the far future that would be seriously cool!

vvaltchev, 2024-02-04:
> If this ever happens in the far future that would be seriously cool!

@PC-Doctor666 do you mean the distro or the Tilck for embedded systems goal ?