vvaltchev/tilck #156: BusyBox build error

turbocat2001, 2023-03-02:
**Describe the bug**
BusyBox build error prevents building the image.

**Host configuration**
 Linux kernel: 5.15.80-1-lts 
 Arch: x86_64
 Distro: Arch Linux

**Reproduction details**
Run:
```bash
 ./scripts/build_toolchain
make
```
Result:
```log
ln: unable to access '/home/max/tilck/toolchain2/i386/busybox/busybox': No such file or directory
```
Although I didn’t notice any problems when I first started the script, and busybox even downloaded.
If you call:
```bash
 ./scripts/build_toolchain --help
```
You can see:
```log
download_gcc_toolchain (*)          [ installed ]
download_cmake (*)                  [ skipped   ]
build_gnuefi (*)                    [ installed ][ i386 x86_64 ]
build_mtools (*)                    [ installed ][ host_x86_64 ]
build_zlib (*)                      [ installed ][ i386 ]
build_busybox (*)                   [ error     ]
```

vvaltchev, 2023-03-02:
Hello @turbocat2001, could you please attach the `toolchain2/i386/busybox/build.log` file ?

vvaltchev, 2023-03-02:
@turbocat2001 I see an error message in Russian:

> /home/max/tilck/toolchain2/i386/busybox/scripts/Kbuild.include:83: *** Рекурсивная переменная «PATH» ссылается сама на себя (в результате).  Останов.

Thanks to Google Translate:

> The recursive variable "PATH" refers to itself (as a result). Stop.

Therefore, the problem seems to be that you have a bad value for `$PATH`. Check it yourself, or copy-paste it here.

vvaltchev, 2023-03-06:
Closing this because it's a bug in user's environment.

closed 2023-03-06 by vvaltchev

turbocat2001, 2023-03-06:
@vvaltchev And again, excuse me, my PATH variable was corrupted. Thanks for putting up with my stupid mistakes.