vvaltchev/tilck #159: Can't build: /scripts/build_toolchain "too many arguments"?

JohnAlexCO, 2023-07-13:
**Describe the bug**
Unable to `build_toolchain` after unzipping project and installing dependencies like `cmake`, `gcc`, `g++`, `clang`...

**Host configuration**
Windows Subsystem for Linux
OS: Ubuntu 20.04 LTS on Windows 10 x86_64
Kernel: 5.10.16.3-microsoft-standard-WSL2
CPU: Intel i5-7200U (4) @ 2.711GHz
GPU: 2ce4:00:00.0 Microsoft Corporation Device 008e
Memory: 72MiB / 6271MiB

gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1 20.04.1)
g++ (Ubuntu 9.4.0-1ubuntu1 20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1

**Toolchain configuration**
Run with no additional configuration or modifications. 
cmake version 3.16.3
GNU Make 4.2.1

**Tilck build configuration**
[2c5b642, Last Week](https://github.com/vvaltchev/tilck/commit/2c5b6421ca91fabfc358a0426f2e86c40fbd9e96), today's version (downloaded July 13 2023)

**Type of issue (select one)**
- Tilck build problem

**Reproduction details**
Downloaded project as *.zip and unzipped project. 
Moved into directory, ran `./scripts/build_toolchain`
```
root@LAPTOP-2GLCKP5T:/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master# ./scripts/build_toolchain
./scripts/build_toolchain: line 36: cd: too many arguments

ARCH=i386
HOST_ARCH=x86_64
GCC_TC_VER=8.4.0
ARCH_GCC_TC_VER=stable-2020.02-2

./scripts/build_toolchain: line 287: /scripts/bash_includes/script_utils: No such file or directory
./scripts/build_toolchain: line 709: /scripts/tc/pkgs/*: No such file or directory
./scripts/build_toolchain: line 686: install_packages: command not found
root@LAPTOP-2GLCKP5T:/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master#
```

**Expected behavior**
`cmake` begins building of dependencies for creating the disk image built using the other scripts.

**Screenshots**
```
root@LAPTOP-2GLCKP5T:/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master# ./scripts/cmake_run
./scripts/cmake_run: line 10: cd: too many arguments
./scripts/cmake_run: line 12: /scripts/bash_includes/script_utils: No such file or directory
[cmake_run] Using '/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master' as build directory
./scripts/cmake_run: line 105: generic_get_version_str: command not found
./scripts/cmake_run: line 106: fail_if_too_old: command not found
./scripts/cmake_run: line 110: generic_get_version_str: command not found
./scripts/cmake_run: line 111: fail_if_too_old: command not found
[cmake_run] Using 'cmake' as CMake executable
./scripts/cmake_run: line 134: cd: too many arguments
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:90 (message):
  Toolchain not found: run scripts/build_toolchain first.

-- Configuring incomplete, errors occurred!
See also "/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master/CMakeFiles/CMakeOutput.log".
root@LAPTOP-2GLCKP5T:/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master# ./scripts/build_toolchain
./scripts/build_toolchain: line 36: cd: too many arguments

ARCH=i386
HOST_ARCH=x86_64
GCC_TC_VER=8.4.0
ARCH_GCC_TC_VER=stable-2020.02-2

./scripts/build_toolchain: line 287: /scripts/bash_includes/script_utils: No such file or directory
./scripts/build_toolchain: line 709: /scripts/tc/pkgs/*: No such file or directory
./scripts/build_toolchain: line 686: install_packages: command not found
root@LAPTOP-2GLCKP5T:/mnt/c/Users/fake pasta/Downloads/tilck-master/tilck-master#
```
I read through that `CMakeOutput.log` and I wasn't able to find anything in it that helped figure out what the problem exactly is.

**Additional context**

lvntky, 2023-07-16:
> @0xdeadbad @lvntky would you like to have a bit of fun with this issue? It's about putting Tilck's main directory in a path with spaces and fix all the places where quotes are not used properly. For example, in this case this:
> 
> ```
> MAIN_DIR="$(cd $SOURCE_DIR/.. && pwd)"
> ```
> 
> I guess should be fixed to be something like:
> 
> ```
> MAIN_DIR=$(cd "$SOURCE_DIR/.." && pwd)
> ```
> 
> But I'm not completely sure it will work perfectly. Maybe this way?
> 
> ```
> MAIN_DIR="$(cd \"$SOURCE_DIR/..\" && pwd)"
> ```
> 
> OR, considering the example of previous line, maybe quotes inside `$()` don't need to be escaped?
> 
> ```
> MAIN_DIR="$(cd "$SOURCE_DIR/.." && pwd)"
> ```
> 
> Of course, @JohnAlexCO you're welcome to propose a fix as well, if you like.

Yes i would like to :) can you assign me please

vvaltchev, 2023-07-16:
Sure, @lvntky. I assigned the issue to you.
Btw, `$( )` creates a new context, so inside it quotes can be used without escaping. Check this link: https://unix.stackexchange.com/questions/118433/quoting-within-command-substitution-in-bash

Therefore, the `MAIN_DIR="$(cd "$SOURCE_DIR/.." && pwd)"` is the correct one. But the point of this issue is not fixing just that particular case, but testing Tilck's build with paths containing spaces and fixing also other places where paths are not quoted properly and break. I guess there are a few places in the CMake files as well where quoting is necessary.

lvntky, 2023-07-24:
Hi sorry for latency i had some issues with my dev computer tonight i try to open a PR :)

vvaltchev, 2023-08-22:
I spent a bit of time checking how big & tricky will be fixing this issue and after fixing 20-30 places I realized that non only it will require quite some effort to support paths with spaces, but overall the code will look uglier with all of those quotes. It's not worth. So I committed ad058400867b79633c3edc070ab4c5061a807eb2 which checks for spaces in the path and fails gracefully with an error if any are found.

closed 2023-08-22 by vvaltchev

Pimpdaddy1274, 2023-10-12:
> **Describe the bug** Unable to `build_toolchain` after unzipping project and installing dependencies like `cmake`, `gcc`, `g++`, `clang`...
>
> **Host configuration** Windows Subsystem for Linux OS: [...]

Pro tip: don't use WSL/WSL2. Just use VirtualBox for God sake.