vvaltchev/tilck #178: Ability to add files to the created image [feature request]

JonathanWilbur, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
I want to create a Tilck distro, and I need to be able to add files, such as shells and compilers, to the `tilck.img` produced from building. I have searched through the scripts for a while now, and it is not clear to me where files get copied in, and how to customize this.

**Describe the solution you'd like**
I think the user-friendly solution would be for there to be a "skeleton folder" that gets copied on top of the image's file system. Other than this, it might be nice to be able to just know / have some documentation on where the files get copied in the shell scripts so it can be easily modified.

**Describe alternatives you've considered**
I know when you build the Linux kernel, you just get a kernel, not a disk image. Would that be possible with this repository? If I just got a kernel executable, I could just configure Tilck as a GRUB menu entry (I think).

**Additional context**
N/A

vvaltchev, 2024-01-04:
Hi @JonathanWilbur, I'm glad that you believe in Tilck so much that you're sure you can make a "Tilck distro", but unfortunately, that's unrealistic at the moment. Tilck is a *small* project and the support for Linux applications is limited. In addition to that, it still has **NO** support for any kind of storage. Therefore, "installing" Tilck doesn't make any sense. Tilck is at the moment just a stateless system that support a limited set of Linux applications. The Linux compatibility is **partial**, not full. And it's not even in the project's roadmap to support everything Linux does. Please read carefully the main README.md file and take a look at all the questions here: https://github.com/vvaltchev/tilck/discussions.

About the skeleton folder, that actually exists and it's the `sysroot` folder. Whatever you put there, will be copied in Tilck's image. But yeah, please don't expect to make an actual distro that way. Supporting each new application requires quite some work and it's not in project's goal to support as much as Linux software as possible. Project's goal is completely different. The Linux compatibility is just a "nice side-feature" of the project. Tilck is NOT an attempt to re-write Linux.

About GRUB, yeah. That's totally possible. Use the multiboot syntax in your GRUB entry and pass the Tilck kernel file (build/tilck) and the fatpart file as a multiboot module. But again, that's something that could be useful only for kernel developers wanted to test Tilck.

At the moment, there are no use-cases for this project outside of the field of kernel development, simply because the project is far from there. One day, I hope to make it useable for embedded systems. Certainly, it won't be become a desktop OS.

JonathanWilbur, 2024-01-04:
Thank you for clearing that up, but that actually still does not disqualify Tilck for what I need, actually! (I think.)

To clarify my use case, I am actually trying to create a hyper-minimalist distro called [PunchcardOS](https://github.com/JonathanWilbur/punchcardos) whose entire codebase is so small that a reasonable human can review _all of it_, and it can be used to compile programs and generate a cryptographic hash over the executables so that other machines that produce executables with the same hashes can trust that no backdoors, malware, telemetry, or other undesirables were slipped into the code or in any of the tools used to build it. In other words, this particular distro is for bootstrapping a trustworthy supply chain of software.

That said, having no persistent filesystem is actually a _feature_ as far as I'm concerned! Ideally, it only really needs the minimal system calls for reading and writing to files (including the terminal / screen, of course).

Unless you still think this is unsuitable, I will close out this issue. Thanks again for this great project!

vvaltchev, 2024-01-04:
Hi @JonathanWilbur. That's interesting!
Well, compiling programs on Tilck is possible with TinyCC at the moment. See `./build_toolchain -h` for the full list of things you can built in the system's image. After you have something in the toolchain, re-run `./script/cmake_run` and then you need to enable it in the CMake file as well. To do that, install cmake-curses-gui on your system and run `ccmake build` in order to enable the EXTRA_TCC option. After that, just run `make` to rebuild the image.

I saw you wanna use another compiler, `chibicc`. While that's possible, note that making a compiler to run is typically a complex task because we're using a cross-compiler toolchain targeting i686 and you need also to configure that new compiler to use the libmusl installed on the Tilck system. Of course, unless you want to also skip any libc in userspace. Making TinyCC to work was tricky. Not sure if that compiler supports i686 or at least i386.

Also, I saw that you would like to send the data over HTTP, SMTP, or FTP. That's cool and it's in Tilck's roadmap to have networking support at some point, but it's not the case yet. At the moment, there is **no** networking support, at all. In theory, you could use the serial ports from Tilck and tell QEMU to redirect their output to a file, but I have to improve the support for serial ports as well. It's not reliable enough at the moment, as I've been focusing on other things.

JonathanWilbur, 2024-01-04:
The ability to use networking is definitely optional, since the point is to ultimately get a hash of a trustworthy executable, which I plan to just write down on the unhackable and ever trustworthy pen-and-paper, then type it out on an untrusted machine with network access (my normal laptop).

I definitely will not be running this in a virtual machine, though, for any purpose other than testing. I do want to remove the underlying host operating system and the hypervisor as a potential vector for surreptitious / unwanted behavior. I do plan to configure the build to be leaner, though.

I didn't realize that Tilck doesn't run on x86_64 though. I'm not sure exactly what that will entail for my project. I imagine most things will work fine, though. I have never helped to develop an operating system, but I would be interested in contributing to this to make it work.

vvaltchev, 2024-01-04:
32-bit Tilck will boot and work on x86-64 machines, I've tested that way, but if we're talking about using it on a HW machine for any practical purposes, that's even harder to get any data in/out the OS. The basic serial port doesn't work on real HW because modern machines don't have legacy ISA serial ports. Therefore, it would require an insane amount of work to get any I/O on a real machine other than just screen and keyboard through PS/2 emulation. I'm not sure that will be enough for any practical purposes.

About contributing to the project, I truly appreciate the desire and feel honored to hear that, but I must warn you that it is very challenging to write an OS. Take a look at the experience required to even begin working on those things according to the OSDEV community:

- https://wiki.osdev.org/Introduction
- https://wiki.osdev.org/Required_Knowledge
- https://wiki.osdev.org/Beginner_Mistakes

Unless you have 10+ experience in C & low-level stuff already, it will take you _years_ to learn the necessary stuff to help. I've invested several times a significant amount of effort in forming new contributors, but when people realized the amount of effort they need to put it to become a kernel developer, they all gave up. Most of the time, people did only super-small things and still found it too challenging. People got a bit of experience and found a job. See the closed pull requests to get a basic idea of what has been done by others.

I guess that the few people that really have the passion to do that, either write their own OS (see the OSDEV community) or they contribute to Linux. I'd be so happy to have a seriously committed contributor, but what would mean working FULL TIME on the project for 1-2 years. Given that this is a no-profit project and there is no compensation, it's hard to find a skilled and motivated person willing to do that. 

Therefore, at least for this project, the best chance I see is for me to continue working on it on my free time, hoping to reach a critical mass and give people something *practically useable* for various projects like your own etc. Only then, I'll probably find serious contributors and be able to scale.

Unless you're either have already some kernel experience or you're willing to spend a very long time working full time on the project and learning, sadly I don't think it's realistic for you to help me. Even if, again, I truly appreciate your kind offer.

I'm really sorry if that's disappointing because Tilck is not ready for any practical purposes at the moment. It's good only if you wanna develop on it, learn kernel engineering etc. I couldn't spend enough time myself to get it to be good for practical purposes, as it takes years to achieve results.

JonathanWilbur, 2024-01-06:
Okay, no problem. I trust your judgement.

As for this issue, I think we can close this issue. I'm sure we'll be in touch again, but hopefully I'll have something cool to show off next time! Thanks for your help!

closed 2024-01-06 by JonathanWilbur

vvaltchev, 2024-01-07:
@JonathanWilbur Thanks for understanding! I'd be happy if you could manage to use Tilck somehow in your project despite its many current limitations, as I understand that's indeed very close to what you're looking for. Hopefully one day it will have all the basic features you need. The thing I'm working now, when I have time and energy during weekends, is the porting to x86-64.