vvaltchev/tilck #180: ./toolchain-wrapper: cannot execute: required file not found

pantantrant, 2024-02-14:
Steps I ran:

./scripts/build_toolchain --skip-install-pkgs
./scripts/cmake_run -DTERM_BIG_SCROLL_BUF=1 -DKERNEL_SHOW_LOGO=0
make -j12
I get this output:

/bin/sh: /root/tilck-master/toolchain2/host/gcc_8_4_0_i386_musl/bin/i686-linux-gcc: not found

Which is actually the command 'toolchain-wrapper' being run with the error message: ./toolchain-wrapper: cannot execute: required file not found

vvaltchev, 2024-02-14:
Hello @pantantrant, I guess that the error is self explanatory: the cross compiler is missing. Also, you run the ‘build_toolchain’ with the option to skill the installation of the system dependencies. Did you at least once run that script without any options first? Also, Tilck has no toolchain-wrapper script. That must be part of your scripts. I cannot debug that.

closed 2024-02-14 by vvaltchev

vvaltchev, 2024-02-14:
@pantantrant I took a 2nd look and searched for `toolchain-wrapper` in Tilck just in case. Yeah, it's a binary in the GCC toolchain, not in my project. It looks like you simply don't have a GCC toolchain downloaded or you deleted it. I don't know.

To be super sure, just delete the whole toolchain2 directory, checkout the repo to the master branch and start over again. Run `./scripts/build_toolchain` without any arguments and let it install what it needs. Then run `./scripts/cmake_run` and so on.

Also note, there are many **good** reasons for me to ask all the questions in the issue template for the bug report: https://github.com/vvaltchev/tilck/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

People do all sort of mistakes and have all the sort of weird system configurations. If you don't answer all the questions, I cannot tell you more.