vvaltchev/tilck #184: Source code in `/sysroot` gets built

JonathanWilbur, 2024-05-11:
**Describe the bug**
I was told that everything in `/sysroot` just gets copied into the bootable image, but it seems as though `make` (or some other tool) is recursively executing Makefiles in these directories.

**Host configuration**
- Docker, in an `ubuntu` container
- `gcc`, although I don't think this is relevant
- x86_64

**Toolchain configuration**
I am actually building this using a Dockerfile: https://github.com/JonathanWilbur/punchcardos/blob/fb6d5b354756ad8c23b774dadab30d509f0a9daa/os3.dockerfile

**Tilck build configuration**
- How did `cmake` was run? Did you use the `run_cmake` wrapper? I used `cmake_run`, here: https://github.com/JonathanWilbur/punchcardos/blob/fb6d5b354756ad8c23b774dadab30d509f0a9daa/os3.dockerfile#L176
  For build errors, please attach the full build log of a *clean* build.
- Tilck commit: `da2964e00e67d008346957a619d579d516cadf24` (cloned today)

**Type of issue (select one)**
- Tilck build problem

**Reproduction details**
- Is the issue reproducible in a deterministic way? Yes.
- Is the issue reproducible both on real hardware and QEMU or just on real HW? Not relevant.
- If applicable, describe the hardware machine used to run Tilck. Not relevant.
- Indicate the exact steps to reproduce the problem:

https://github.com/JonathanWilbur/punchcardos/blob/fb6d5b354756ad8c23b774dadab30d509f0a9daa/os3.dockerfile#L135-L138

If I uncomment the last line (which extracts the downloaded archive), this fails.

**Expected behavior**
I didn't expect `make` to recurse into files in `/sysroot`. I would expect nothing to be touched in there, other than for binaries to be copied in as they are built elsewhere.

**Screenshots**

```
21.81 [ 98%] Linking CXX executable fathack
21.85 [ 98%] Built target fathack
21.86 [ 98%] Creating a dual-bootable (legacy+efi) image file (MBR)
22.05 stage0/stage2/High_level_prototypes/lisp: No such file or directory
22.05 make[2]: *** [CMakeFiles/tilck_image.dir/build.make:94: tilck.img] Error 1
22.05 make[2]: *** Deleting file 'tilck.img'
22.06 make[1]: *** [CMakeFiles/Makefile2:319: CMakeFiles/tilck_image.dir/all] Error 2
22.06 make: *** [Makefile:91: all] Error 2
```

As seen above, it seems to recurse into the `stage0` repository which I download and extract into `/sysroot` and start building it. There may be a problem with `stage0`, but I don't think the build should interfere with files put in `/sysroot` by users.

vvaltchev, 2024-05-11:
Hello @JonathanWilbur, 
this issue seems to be entirely related with your scripts and it’s not fair asking me to debug your build to understand what’s going wrong. 

For more info about how the files from sysroot and other stuff is copied into the fatpart file, check the build_fatpart script. Briefly, it first copies the contents of the sysroot directory from the source tree to the build directory, then it copies there all the built binaries, then it gets the  list of all the files and directories there using ‘find’ command and uses mcopy to copy each file to the fatpart file/fat32 partition. 

Therefore, before opening an issue, please try to isolate the problem. Open a clean repo of Tilck and put in sysroot one file. No docker files, no fancy logic. Does it work? If it does, very likely the issue has nothing to do with this project. 

Cheers,
Vlad

closed 2024-05-11 by vvaltchev

JonathanWilbur, 2024-05-12:
Hey Vlad,

I am not asking you to debug my build, which--by the way--does nothing more than build the binaries that your build script uses from source, places them in the expected locations, then runs your build script. I figured if I described the problem, you might know right away whether or not your build recurses into the `/sysroot` directory and builds stuff in any folder that has a `Makefile`. If you agreed this was indeed the case, we could probably agree this would be a bug, and if you knew this not to be the case, I would happily investigate further.

That aside, I did as you asked (except that I run these build steps in a Dockerfile, still, but that is _more_ reproducible than me running them on my bare machine). This exact Dockerfile fails with the exact same problem:

```dockerfile
FROM ubuntu:latest
RUN apt update && apt install -y git git-core gcc g++ gcc-multilib \
    g++-multilib clang cmake python3-minimal python3-serial python3-pip apt \
    gawk unzip parted wget curl qemu-system-x86 imagemagick
RUN git clone --depth 1 https://github.com/vvaltchev/tilck.git
WORKDIR /tilck
# See: https://github.com/vvaltchev/tilck/issues/177
# We don't want any of the interactive features of apt, so we just disable them.
ENV DEBIAN_FRONTEND=noninteractive
ENV RUNNING_IN_CI=1
RUN mkdir -p /tilck/sysroot/stage0
WORKDIR /tilck/sysroot/stage0
RUN curl -LO https://github.com/oriansj/stage0/archive/refs/tags/Release_0.4.0.tar.gz
RUN tar --strip-components 1 -xvzf Release_0.4.0.tar.gz
WORKDIR /tilck
RUN ./scripts/build_toolchain
RUN make
```

Also, I did look in `build_fatpart` and I did not notice anything that seemed to recursively `make` the `/sysroot` directory, but I am not confident that I didn't just miss it.

vvaltchev, 2024-05-12:
Hello @JonathanWilbur,
Tilck's build does not "recurse" into the sysroot directory, at all. The only thing that happens is that the `build_fatpart` script is run: https://github.com/vvaltchev/tilck/blob/master/scripts/templates/build_fatpart.

The error:
```
22.05 stage0/stage2/High_level_prototypes/lisp: No such file or directory
```

has *nothing* to do with `make` entering ("recursing") into the `sysroot/` directory. What probably happens is that you have whitespaces in the paths and therefore the `build_fatpart` script doesn't work as expected. See:

```
   # first, create the directories
   for f in $(find * -type d); do
      $mmd -i $dest $f
   done
```

and:

```
   # then, copy all the files in sysroot
   for f in $(find * -type f); do
```

From the `build_fatpart` script. Clearly, that won't work if the paths have whitespaces, but that's expected. I added a nice check in the `build_toolchain` script for whitespaces in the path containing Tilck's source. Unfortunately, your case is different because you're adding new paths into the sysroot/ directory. 

Did you check if such a file `stage0/stage2/High_level_prototypes/lisp` was supposed to exist ? 

The Tilck build does **not** support whitespaces in ANY path simply because it's very ugly to support reliably: requires escaping everywhere in multiple levels sometimes, plenty of extra tests and simply it's not worth doing so. The code will become uglier and harder to read, while the benefit would be very small. I don't believe supporting paths with whitespaces will bring huge benefits to this project.

This is probably a duplicate of: https://github.com/vvaltchev/tilck/issues/159

Cheers,
Vlad

vvaltchev, 2024-05-12:
@JonathanWilbur I reproduced this in 5 seconds. I added a file called "file with whitespaces" to sysroot/ and run a build. It failed like this:

```
Consolidate compiler generated dependencies of target fathack
[100%] Built target fathack
[100%] Creating a dual-bootable (legacy+efi) image file (MBR)
file: No such file or directory
make[3]: *** [CMakeFiles/tilck_image.dir/build.make:99: tilck.img] Error 1
```

Complaining that "file" doesn't exist, while that's only the first word of its name. So, was "lisp" supposed to be a filename of just the first word of a file with whitespaces?

vvaltchev, 2024-05-12:
@JonathanWilbur I added a check to make sure that you'll get a graceful error next time. See the latest commit in the master branch.

Btw, if you allow me, I'm a bit concerned about your "PunchcardOS" project and the idea of "start with Tilck as a lighter alternative to Linux". This kernel is not an alternative to Linux and trying to make things work with Tilck will be 100x harder than using directly Linux, simply because Tilck is very limited in features and only partially compatible with Linux. Almost *nothing* will work out-of-the-box except for the few things that already do work. In particular if your plan is to use Tilck as a starting point and later move to Linux, that doesn't make sense to me. Tilck is the "hard way", Linux is the "easy way".

Making Bash to work on Tilck will require a significant effort on the kernel side and I won't have the time to do that. Same thing for GCC and most of the other stuff that you mentioned in your README file. It won't help if you opened an issue requesting me to implement that feature, because I really have no time for that, due to my full-time job. And when I have a bit of time and energies to work on this project, I do other stuff like porting it to x86-64, fixing subtle bugs etc. It's not in my roadmap to support all of that and even if it was, it would take me years to complete.

I understand and appreciate your idea of using this project for building something else. In principle I'm glad that someone wants to do anything with it, given how limited it is. However, I'm trying to be realist here and point out that you are certainly going to hit very serious walls while trying to achieve what you aim, if you use Tilck. Using it as a temporary/simpler solution instead of Linux is definitively the wrong approach. It would only make sense if you wanna stick with Tilck forever. In that case, you'd really need to spend *years* trying to become a kernel developer because operating systems are complicated. In that case, very slowly you'll be able to make the contributions necessary to Tilck to make your project work the way you need, when that doesn't break my design and goals for the project. But.. you'd be spending 95% of your time on Tilck and 10% on your project if you do that. 

Think about it. I just wanna set the right expectations for you given the walls you're going to hit while trying to run other software on Tilck. It probably won't work, and I won't have the time to fix it for you.

Cheers,
Vlad

JonathanWilbur, 2024-05-12:
Oh huh, I see now that there is a single space after the word `lisp` in this file name, which I only noticed when pulling up the file in my file browser. That explains it. And I'll take your word for it. Thank you.