vvaltchev/tilck #227: Tilck on RISCV64 crashes with NULL pointer dereference when UBSAN is enabled

vvaltchev, 2024-09-16:
Hello @aenrbes,
I re-worked Tilck's test infrastructure and created an Azure pipeline to run also all the tests for Tilck RISCV64.
However, I discovered multiple issues. In some case (e.g. RELEASE builds) the kernel couldn't even build for RISCV64.

I fixed some of the issues like the build-related ones, but I'd like to ask for help for the remaining ones, in case you can do that. In order for me claim full support for RISCV64, the new architecture need to pass successfully the same kind of tests the older one does. It might be limited in features, but the correctness and quality should be on par with the other one.

For example, when the option KERNEL_UBSAN is enabled, the `getrusage()` test crashes like this. I suspect that the problem is related with the RISCV64 implementation of `fault_resumable_call()`. Can you check that?

```
[devshell] [RUN   ] getrusage

********************** KERNEL PANIC **********************
UBSAN: null pointer dereference
UBSAN: /home/vlad/dev/tilck2/common/string_util.c:265:14
Current task [USER]: tid: 252, pid: 252
Interrupts: [ 8 ]
sstatus: 0x0000000000040022 [ SIE SPIE SUM ]
ra: 0x000000200020df46, sp: 0x000000203804c758, gp: 0x000000000004a420, tp: 0x000000000004e980
t0: 0x0000002007f58800, t1: 0x0000000000000000, t2: 0x0000002000206d58, s0: 0x000000203804c778
s1: 0x0000000000000000, a0: 0x0000000000000000, a1: 0x0000000000000000, a2: 0x0000000000000200
a3: 0x000000200021b314, a4: 0x000000203804c810, a5: 0x0000002000320110, a6: 0x000000200026b76a
a7: 0x00000000000000a5, s2: 0x0000000000000109, s3: 0x000000203804ca58, s4: 0x0000000000000fa0
s5: 0x000000203804c8d8, s6: 0x0000000000048ff0, s7: 0x000000000003cf50, s8: 0x000000000003ba08
s9: 0x000000000003bc00, s10: 0x000000000003d468, s11: 0x0000000000049870, t3: 0x0000000000000075
t4: 0x0000000000000072, t5: 0x0000000000000074, t6: 0x0000000000000065, sepc: 0x0000000000032530
sstatus: 0x0000000000040022, sbadaddr: 0x0000000000000000, scause: 0x0000000000000008
usersp: 0x0000000000000000
Stacktrace (11 frames):
[0x000000200020e1a2] panic + 0x454
[0x00000020002758cc] __ubsan_handle_type_mismatch_v1 + 0xe0
[0x000000200021865a] memcpy + 0x70
[0x0000002000206d0c] fault_resumable_call + 0xd2
[0x000000200027603e] copy_to_user + 0x4c
[0x000000200026b948] sys_getrusage + 0x1de
[0x0000002000201420] do_syscall_int + 0x138
[0x0000002000201de6] do_syscall + 0x29c
[0x000000200020213c] handle_syscall + 0xea
[0x000000200023fd5c] syscall_entry + 0xc2
[0x000000200020fec8] asm_trap_entry + 0xc0

System memory map from multiboot:

           START                 END        (T, Extr)
00) 0x0000000080000000 - 0x000000008001ffff (2,     ) [     128 KB]
01) 0x0000000080020000 - 0x00000000800fffff (1,     ) [     896 KB]
02) 0x0000000080100000 - 0x00000000801fffff (2, LMRS) [    1024 KB]
03) 0x0000000080200000 - 0x00000000803aafff (2, KRNL) [    1708 KB]
04) 0x00000000803ab000 - 0x00000000815fffff (1,     ) [   18772 KB]
05) 0x0000000081600000 - 0x000000008172afff (2, RDSK) [    1196 KB]
06) 0x000000008172b000 - 0x0000000087ffffff (1,     ) [  107348 KB]

Halting the system...
System halted.
```

aenrbes, 2024-09-17:
@vvaltchev 
    I looked at the test code for getrusage and found a test that passed a null pointer to the getrusage call as follows, 
```
  // Check failure when the provided buffer pointer
   // doesn't refer to an accessible page
   void *bad_ptr = NULL;
   rc = getrusage(RUSAGE_SELF, bad_ptr);   =====================> passed a null pointer
   DEVSHELL_CMD_ASSERT(rc == -1 && errno == EFAULT);

   printf("OK\n");
   return 0;
```

So I looked at the code for` sys_getrusage()` and found that there was no special treatment for NULL Pointers, which were passed directly to `copy_to_user()` as follows. So as far as I'm concerned the `__ubsan_handle_type_mismatch_v1` warning that is triggered here is the expected behavior. 
Am I missing something on this question, or am I misinterpreting it?
```
if (copy_to_user(user_buf, &buf, sizeof(buf)))
    return -EFAULT;
```

vvaltchev, 2024-09-17:
Hi @aenrbes, thanks for taking a look!
Yeah, the test intentionally passed a NULL pointer to the syscall to check that it will behave correctly and return -EFAULT. The `copy_to_user()`  call is supposed to be robust against that, thanks to the fault_resumable_call. On i686, UBSAN doesn't catch anything in this specific case, and we trigger an actual fault, from which we resume and make `copy_to_user()` to fail. Somehow on RISCV64 we behave differently. I hope that your theory is correct and that we just need to fool the compiler to perform the actual NULL pointer deference instead of triggering an UBSAN warning. I was concerned about a subtle bug with the assembly code of fault_resumable_cal itself.

aenrbes, 2024-09-18:
@vvaltchev 
I did some research into the behavior of `copy_to_user()` on the i386 platform, and found that the reason why i386 does not trigger a ubsan warnings for NULL pointers is because the i386 `memcpy()` is an inline assembly function.
If I replace all the inline assembly functions in `asm_x86_strings.h` with the C language version in `string_util.c`, `copy_to_user()` on the i386 platform will also trigger a ubsan warning with a NULL pointer.

vvaltchev, 2024-09-18:
Makes perfectly sense. I forgot that we don't have an inline asm implementation of memcpy() on RISCV64, yet.
Yeah, I also agree with the NULL pointer check in copy_to_user(). I hope that UBSAN won't try to be any smarter than that :-)

closed 2024-09-18 by vvaltchev